ess-dmsc/nexus-constructor #27: Windows executable

matthew-d-jones, 2018-10-01:
Having a single file executable or installer early in development would make life easier for potential testers of the application.
`pyinstaller` and `pyupdater` are probably the standard tools for this at the moment, worth checking though.

ACLay, 2018-10-03:
PyInstaller has a [known bug](https://github.com/pyinstaller/pyinstaller/issues/3689) that's preventing it building our demo QML app, so I'm looking into alternatives at the moment.

matthew-d-jones, 2018-10-03:
If there is no obvious alternative then waiting for the PR to be merged to fix that bug is also an option.

ACLay, 2018-10-03:
There are [a few other options](https://docs.python-guide.org/shipping/freezing/) to at least try out.

ACLay, 2018-10-03:
Adding that branch `git+https://github.com/tobinjones/pyinstaller.git` to requirements does seem to fix the initial bug. It does however reveal another pysinstaller-pyside2-QML specific bug, which I think is the one that was guessed to exist [here](https://github.com/pyinstaller/pyinstaller/issues/3711#issuecomment-418678385) based on a similar bug for pyinstaller-PyQt5-QML.

However, [cx_freeze](https://pypi.org/project/cx_Freeze/) is looking very promising, already have a slightly modified version of the program running in it.

PR #35 (2018-10-04): Enable distributable versions of the app to be built
Closes #27 

The app, tests, and distributable version all seem to be good on my windows machine.

The build tool it uses is cross platform, so @matthew-d-jones would you be able to check if the build distributable command (`python setup.py build_exe`) works on your linux box?
closed 2018-10-04 by matthew-d-jones