ess-dmsc/nexus-constructor #52: Improve memory useage in repeated mesh detector graphics buffers

ACLay, 2018-11-15:
Currently, the QtOFFGeometry class repeats the mesh for each instance in a grid.

This means the graphics buffers use much more memory than needed, directly proportional to the number of instances in the grid. Instead, it should just create the mesh for a single instance, and a model should create unique transform matrices for each pixel instance that can be used in the QML.

ACLay, 2018-11-22:
That said, if we had a very simple pixel model, say a square in a 100x100 grid, this proposed change would reduce the graphics memory, but require the CPU to calculate 10,000 transform matrices instead of 1. This may not be as effective a boost as initially thought for real applications...

rerpha, 2019-07-05:
Let's get the 3d view with transformations working in the widgets code first. I believe instanced rendering is coming in the next couple of releases of Qt3D so let's wait until there's a nice API for it. Closing this for now.

closed 2019-07-05 by rerpha