ess-dmsc/nexus-constructor #81: Add a PR Jenkins job to build the distributable on different platforms

matthew-d-jones, 2019-02-06:
Windows, Mac and Linux distributables which should be downloadable as build artifacts after the Jenkins job is complete.

rerpha, 2019-02-07:
blocked by #97

rerpha, 2019-02-18:
Might have a play around with PyInstaller instead of cx_freeze at some point and see if we can get the builds working on all platforms. cx_freeze seems to be including some paths that will never exist outside of jenkins and therefore the executables fail to run

matthew-d-jones, 2019-02-19:
> Might have a play around with PyInstaller instead of cx_freeze at some point and see if we can get the builds working on all platforms. cx_freeze seems to be including some paths that will never exist outside of jenkins and therefore the executables fail to run

Alex originally looked into using PyInstaller, he encountered a couple of bugs with using it with pyside2, see https://github.com/ess-dmsc/nexus-constructor/issues/27
I think creating an appimage would be the the alternative approach to look into.

PR #89 (2019-04-03): Creating an executable for Windows and Linux in Jenkins pipeline
Creating an executable using a parallel build step in the Jenkinsfile 

Disabled the mac one for now and created #88 

Closes #81
closed 2019-04-03 by rerpha