ess-dmsc/nexus-constructor #88: support Mac OS and create build in jenkinsfile

rerpha, 2019-02-06:


rerpha, 2019-04-04:
most of the code is already in the Jenkinsfile, just not enabled. This should just be a case of enabling the mac build.

SkyToGround, 2019-05-08:
I tried to build a Mac binary on my machine and it failed of course. I played around with it a bit without any progress. Searching the web, I found no clear reason as to why it is not working but some suggestions are:

- Use PyInstaller (instead of cx_Freeze)
- Use PyApp builder (instead of cx_Freeze)
- Use Homebrew for supplying Python
- Do not use the system version of Python
- Do not use a version of Python provided by a packaging system

In short, I can live with this not working on MacOSX right as it is such a pain to get to work. We/I should get back to the issue when we start having users.

rerpha, 2019-06-19:
I think we should close this for now. We are going to stop building executables on windows anyway and i think running tests on the mac node might be pointless. 

What does everyone else think?

closed 2019-06-24 by rerpha