ess-dmsc/nexus-constructor #142: Design review

matthew-d-jones, 2019-03-18:
As the scope of the application has changed considerably and will now involve a much larger resource investment and possibly involvement of other institutions, Dominic wisely proposes doing a design review of the application at this stage.

Suggested output:
- Documentation of the current design
- Documentation of proposed design
- Tickets for proposed changes

Week of 1 April would be good if possible. Most of us will be away at the ECP meeting the week before and ACCU the week after.

DominicOram, 2019-03-20:
We did the design review. Created the tickets 152, #153, #154, #155, #156, #157, #158, #159

Can someone please review by:
* Confirming that I captured everything out of the meeting
* Confirming that the above tickets are well written

matthew-d-jones, 2019-04-04:
Thanks. The tickets make sense to me and there is nothing missing from the notes I made, so I'll close this issue.

closed 2019-04-04 by matthew-d-jones