ess-dmsc/nexus-constructor #155: ComponentType should be an enum in the QML

DominicOram, 2019-03-20:
Although a ComponentType is an enum in the python it is a string in the QML, this leads to various casts and checks being performed. It looks like we might be able to register an enum like [this](https://www.riverbankcomputing.com/static/Docs/PyQt5/metaobjects.html#q-enum-and-q-flag). This ticket is to:
* Move ComponentType into a better enum
* Add unit tests where appropriate
* Document the use of enums in QML/python

rerpha, 2019-03-22:
I don't think Q_ENUM is implemented in pyside2 yet going off of [this issue](https://bugreports.qt.io/browse/PYSIDE-957)

rerpha, 2019-06-14:
This is no longer an issue as we're using a dictionary of nexus base classes in #307

closed 2019-06-14 by rerpha