ess-dmsc/nexus-constructor #157: Transform objects should know their parents and children

DominicOram, 2019-03-20:
Currently the information about the order that Transforms are performed is contained within the Transform_Model. There are a number of issues with this:
* It is unclear on the relationship between components and transform models
* The logic inside transform_model is complicated

It was proposed during #142 that the logic of parent/child transforms should instead be contained within the transformation objects themselves. e.g. each will hold a reference to it's parent and children. Components can then hold references to transformations and there will not be a need for a class dedicated to knowing about order.

Edge cases that should be thought about with this are:
* Transformations do not need to be within a component
* What happens when a transform is removed? Are child transforms rerouted or do we just not allow this?

This ticket is to:
* Better understand the logic within transform_model
* Confirm that the proposed data structure better suits the problem and covers all use cases
* Implement the solution
* Add unit tests where appropriate
* Document the data structure

rerpha, 2019-04-25:
- The link between components and transforms is a list index - seems very fragile. 
- We currently don't allow components to be deleted if any transform relies on the component

- The transform model contains a list of all transforms - these transforms are accessed by index from QML (and python) 
- The transform model DOES NOT contain a list of the parent transforms - this logic is handled by the component itself. (it really shouldn't be handled by the component if the transforms are not the component's responsibility)

essentially, the list in `InstrumentModel` compares against the list in `TransformationModel` to get the transforms for each component.

rerpha, 2019-04-25:
I think we could use something like this: 

https://anytree.readthedocs.io/en/latest/

We could use a tree for the transforms, with each branch linking the dependent and parent transforms. 

In each node we could store the transform and optionally the component (as not all components have transforms, and not all transforms have components)

rerpha, 2019-04-25:
alternatively we write our own tree, something like this 

```python

class TransformList:
    def __init__(self):
        self.head = None

class Node:
    def __init__(self):
        # references - do not take ownership
        self.head = None
        self.tails = []
        self.component = None

        # do take ownership of the transform
        self.transform = None

    def has_dependent_transforms(self):
        return not self.tails

    def has_parent_transforms(self):
        return self.head != False
```

DominicOram, 2019-04-26:
Do we also need the component to have a reference to it's transforms as well as the transform have a reference to the component?

rerpha, 2019-04-29:
> Do we also need the component to have a reference to it's transforms as well as the transform have a reference to the component?

I'm not sure how easy that would be. I think you're right, but I don't know how you could link the two together from within a tree unless you gave the component a list of nodes and the transform a list of components

rerpha, 2019-05-07:
Have set up a design review for this for 10/05/2019 

here are some notes in preparation for it: 
- `InstrumentModel` contains a list of components, and a separate list of `TransformModel`s
- `TransformModel` contains one single transformation, ie one rotation 
- `Component`s in the `InstrumentModel` also have a list of transforms and a parent transform - they also seem to have a `dependent_transform` field which looks like it's just one transform. This is a problem because there could be multiple dependent transforms for one transform. Both of these variables come from `TransformControls.qml` which instantiates/stores a TransformModel.

There is some absolutely diabolical maths going on which is completely undocumented for getting transformations to the component from the list of `TransformModel`s and I think this responsibility should be on something other than the `InstrumentModel`.

rerpha, 2019-05-20:
The outcome of this meeting was that we need a tree-like structure for transformations where:
- Transformations are stored in each node in the tree
Each node is comprised of: 
- Parent transform
- Component to apply to 
- Transform details/object
- List of child transforms

The component has a list of these nodes. Note that a single transform can only affect one component, but components can have several transforms.

rerpha, 2019-06-19:
I think @SkyToGround is doing this in his listview branch?

rerpha, 2019-07-01:
Closed in #350

PR #350 (2019-07-01): Transforms know their dependents
### Issue

Closes #157

### Description of work

Keep track of what transforms and components depend on a given transform by storing this information in an `dependee_of` attribute in the transformation. This is then used to stop a transformation being deleted if anything else is still dependent on it. This is covered by unit tests in `test_component.py`.
As `dependee_of` is not part of the NeXus standard we cannot rely on it being present in any existing NeXus files which we load, so I make sure it is generated on file load. This is covered by a unit test in `t [... cut]
closed 2019-07-01 by rerpha