ess-dmsc/nexus-constructor #171: Some other useful QML-related things to add to the wiki

DolicaAkelloEgwel, 2019-03-22:
- functions
- Singletons and `qmldir` for creating variables that can be shared across files
- Row/ColumnLayout
- aliases
- Explanation of the purpose of the custom types: `LabeledTextField` and `ActiveScrollBar` (more for dev notes rather than the QML tutorial bit)
- QVariant
- anchors

@rerpha @DominicOram Do you think these should all be in the wiki or is it not needed? Also what else would be useful?

rerpha, 2019-03-22:
Definitely should be in the wiki, I think we should jot down all we know about QML :+1:

rerpha, 2019-04-24:
- [ ] functions
- [ ] singletons and qmldir
- [ ] row/columnlayout
- [ ] aliases
- [ ] Explanation of the purpose of the custom types: LabeledTextField and ActiveScrollBar (more for dev notes rather than the QML tutorial bit)
- [ ] QVariant
- [ ] anchors

rerpha, 2019-05-30:
closing this as we are switching to widgets.

closed 2019-05-30 by rerpha