ess-dmsc/nexus-constructor #187: Test distributable packages

matthew-d-jones, 2019-04-08:
Either automated test that it runs, stops and checks exit code or at least include in documented steps to create a release.

rerpha, 2019-04-18:
we might run into this issue when running the nexus-constructor headless - 

```
root@9e07cedb58c8:/nexus-constructor/build/exe.linux-x86_64-3.6# ./NexusConstructor 
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)
```

rerpha, 2019-11-01:
done in #479

PR #479 (2019-11-01): fix executable and windows build
### Issue

Closes #187 

### Description of work

Unpinning cx_freeze - the issue on windows has been fixed in 6.0
https://github.com/anthony-tuininga/cx_Freeze/issues/407

Also fixing the executable (had been broken for a while) and developing a test to make sure it doesn't break again. 

code changes mean that the definitions dir only has to be found once rather than all over the code.

### Acceptance Criteria 

N/A

### UI tests

N/A

### Nominate for Group Code Review

- [ ] Nominate for code review
closed 2019-11-01 by matthew-d-jones