ess-dmsc/nexus-constructor #218: Modification of a component should trigger a live update of JSON and 3D model

SkyToGround, 2019-05-01:
Currently, when editing a text field, you have to press tab or enter to update the 3D view and JSON code. This should instead be done automatically while editing. If it is not desirable to update with every key-stroke, a timer can be used to trigger the update.

rerpha, 2019-05-03:
You can also click away from the text box. 

The issue with this is our code is too slow to keep regenerating the mesh each time. #100 should solve this but i have no idea of how to implement it. I think the bottleneck is how we're transforming outside of Qt - Qt uses OpenGL in the backend so should be very fast but our code doesn't use any kind of parallelism/concurrency for doing that sort of thing.

SkyToGround, 2019-05-03:
@rerpha As mentioned above, if the code is to slow for an update at every key stroke there are alternatives. The method I have been using in other projects is to set a timer of (for example) 1 second on the first key stroke that will trigger an update after that time has passed. Any subsequent key strokes that happen before the timer has triggered the update callback function will not generate a new timer.

rerpha, 2019-07-16:
Qwidgets line edits should trigger an action on their textChanged signal. I am going to close this for now as it should happen when the transformations stuff works.

closed 2019-07-16 by rerpha