ess-dmsc/nexus-constructor #219: Failure to parse numeric values with decimals

SkyToGround, 2019-05-01:
It appears that the widgets that parse numeric values can not properly handle decimal values on machines that are not set to use English language settings.
E.g. the application properly determines that the "," character is the decimal delimiter used in the Swedish language. However, the string ``"12.12"`` is turned into the numeric value ``12``.

Alternative 1: Remove support for international representation of numeric values.
Alternative 2: Fix international representation of numeric values.

rerpha, 2019-05-08:
https://www.qtcentre.org/threads/63501-comma-as-decimal-seperator

kmurica, 2021-02-23:
I believe alternative 2 would be quite easy to fix for this issue.

closed 2022-01-04 by kmurica