ess-dmsc/nexus-constructor #221: Rendering issue caused by hiding JSON view

SkyToGround, 2019-05-01:
Re-production steps:
1. Disable the JSON view.
2. Minimise window.
3. Re-enable JSON view.

Fix alternative 1:
Change minimum width of 3D view when switching off/on the JSON view.

Fix alternative 2:
Change window size (if required) when switching on JSON view.

matthew-d-jones, 2019-05-02:
This is not observable on Linux (or at least on Ubuntu KDE 18.04).

SkyToGround, 2019-05-03:
@matthew-d-jones I was maybe not very clear in my re-production steps. Step 2 should be: resize the window to its minimum width.

matthew-d-jones, 2019-05-07:
I don't notice anything wrong in the window when I do that, but I do get a couple of these in the console:
```
QOpenGLFramebufferObject: Framebuffer incomplete attachment.
```

PR #249 (2019-05-07): Rendering issue when hiding the JSON pane and resizing the window to minimum
### Issue

Closes #221 

### Description of work

Setting a minimum width for the JSON pane so the view never goes out of display 

there are still QOpenGL errors, but im not sure how to fix these, might be a bug in QML? 

### Acceptance Criteria 

- hide json pane 
- Resize the window to as small as possible
- re enable json pane
- check that window is resized again

### UI tests

Not changed

### Nominate for Group Code Review

- [ ] Nominate for code review
closed 2019-05-07 by rerpha