ess-dmsc/nexus-constructor #224: Unit in cylinder component "Full editor" is ignored

SkyToGround, 2019-05-01:
It is not possible to set this to anything other than "m". When re-opening the editor, the value (unit) is reset to "m".

matthew-d-jones, 2019-05-02:
I suspect the units are only being used when a CAD file is loaded. We'll need to store them in the model so that the setting persists.

matthew-d-jones, 2019-05-28:
This turns out to be non-trivial to solve in a nice way, due to QML. As we are planning to move to widgets we should revisit this ticket once that is complete.

rerpha, 2019-07-05:
Closing this as it only affects the QML code. 

This should be sorted out in #322 as that should just use the "add component" window in which the units line edit can be edited before or after adding the geometry info.

closed 2019-07-05 by rerpha