ess-dmsc/nexus-constructor #270: Should we switch to using Qt widgets rather than QML?

rerpha, 2019-05-10:
Qt widgets are still being developed, and QML doesn't seem to play as nicely with pyside2. I think we should switch to widgets, but we need to do some research first: 
- investigate how long it would realistically take to make the switch to widgets 
- see if we can do this gradually and use a mixture of the two? 
- see what other libraries use - silx, nicos etc. 
  - NICOS uses pyqt5
  - silx supports both bindings so either would work
- investigate if high dpi displays are an issue with widgets - QML has this advantage if so

We could either keep using pyside2 or switch to pyqt5, or develop on qtpy(wraps pyqt5 using the pyside2 API) 

There seems to be better support for testing widgets in the way of UI tests and unit tests, pytest-qt seems to only work with widgets so we would be able to automate those. The logic in python can be unit tested easily, as can the QML objects. 

Widgets in theory would be much quicker to develop - they are much easier to refactor and there are free tools available to help design the UI. (Qt designer) 

Pyside2 is still being developed but pyqt5 has been around longer and has more documentation - however Qt for python(using the pyside2 module) is now officially supported by the Qt company. 

We should have a discussion on this and anything else that we need to consider.

rerpha, 2019-05-10:
this might solve the DPI issue but we'll need to see: https://coad.ca/2017/05/15/one-way-to-deal-with-high-dpi-4k-screens-in-python/

rerpha, 2019-05-13:
https://build-system.fman.io/

look like there's some nice packaging tools for pyqt5

matthew-d-jones, 2019-05-23:
Decisions made in meeting on 2019-05-20:
- Stick with PySide2
- Proceed with prototyping the main window, component/transformation listview, add component dialogs and 3D view with widgets.

This should allow us to then replace the entire interface with widgets in one go to avoid the problem of getting models between Python and QML.

closed 2019-05-23 by matthew-d-jones