ess-dmsc/nexus-constructor #318: Widgets: generate definition html files from definitions submodule instead of grabbing from web

rerpha, 2019-06-13:
This means we don't rely on an internet connection to run the NeXus constructor.

DolicaAkelloEgwel, 2019-07-03:
Converting the XML files to HTML seems to be pretty involved without the help of make. Looks like this is being put on hold for now.

rerpha, 2019-07-16:
I wonder if we should just build the packages every so often manually, commit them and keep them inline with the submodule. building them every time you run the program will take ages, and we can't have make as a requirement for the nexus constructor very easily. Seeing as they don't change very frequently, we could just build them every time we change the submodule and keep them in a html directory?

rerpha, 2019-10-10:
this isn't as straightforward as it seems and given that it's working perfectly fine currently and we've never had any issues with it i am going to close this ticket.