ess-dmsc/nexus-constructor #339: Deal with units in Transformation

matthew-d-jones, 2019-06-27:
Add a units getter and setter (`@property`) to `Transformation`.
We keep everything in SI in the application, so getter should convert from whatever is in the file to SI using the `pint` library. Setter can assume it is given SI units, but this should be clearly documented in its method description comment.

zjttoefs, 2019-06-27:
Are you sure? The SI unit for angle is radian.

matthew-d-jones, 2019-06-27:
@zjttoefs Yes, this would be for the 3D view to calculate where to display things, not for how the transformation's numerical value is displayed to the user.

rerpha, 2020-02-18:
This should be much easier once #632 is merged as it adds the unit field to transformation along with a convenient way to edit it

rerpha, 2020-03-06:
`units` are now in the transformation class so this should just be a case of tying them up with the transformation magnitude.

umithardal, 2021-02-26:
The description is unclear. The tickets needs a second look.

closed 2021-11-09 by mattclarke