ess-dmsc/nexus-constructor #375: Make it possible to edit components

SkyToGround, 2019-07-09:
The `ComponentInfo` item should be used to edit some things (e.g. the component name). It should also have a button for opening a window that allows detailed editing of a component.

matthew-d-jones, 2019-07-09:
Ticket for editing Component is here: #322
Being able to edit component name from the component treeview may be a good idea, but we should be cautious about adding too much complexity there from a usability perspective. Given that the functionality would be present in the edit component window, I suggest we start using the software first so that we discover whether renaming a component is a common thing to want to do.

rerpha, 2019-07-29:
Going to mark this as lower priority for now as it is possible to edit the name from the "edit component" window.

rerpha, 2019-11-01:
For simplicity's sake, let's just leave the edit component window to do this job rather than having two edit fields that could cause conflicts. I am going to close for now unless anyone disagrees.

closed 2019-11-01 by rerpha

SkyToGround, 2019-11-01:
I am ok with closing this for now.