ess-dmsc/nexus-constructor #377: Fix crash on exit

SkyToGround, 2019-07-09:
The application crashes on exit on my machine. It seems likely that this is caused by some python resource is being freed before the C++ Qt backend is done with it. Though currently only a nuisance, it should be fixed.

rerpha, 2019-07-09:
this is the stack trace i get when exiting: 

```
Warning: Setting a new default format with a different version or profile after the global shared context is created may cause issues with context sharing.
[Qt3DRender::GLTexture] No QTextureData generated from Texture Generator yet. Texture will be invalid for this frame
Traceback (most recent call last):
  File "/home/jack/code/repos/nexus-constructor/nexus_constructor/neutron_animation_controller.py", line 34, in set_distance
    def set_distance(self, distance):
KeyboardInterrupt
```

looks like it's coming from `neutron_animation_controller.py`

DolicaAkelloEgwel, 2019-07-09:
This is what I get from gdb python
```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x0000000000650068 in ?? ()
(gdb) bt
#0  0x0000000000650068 in ?? ()
#1  0x00007fffd34e8c19 in QMetaObject::cast(QObject const*) const ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt5Core.so.5
#2  0x00007fffb6f8d75b in Qt3DCore::QNodePrivate::notifyDestructionChangesAndRemoveFromScene() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt53DCore.so.5
#3  0x00007fffb6f8dcc7 in Qt3DCore::QNode::~QNode() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt53DCore.so.5
#4  0x00007fffb6f92979 in Qt3DCore::QEntity::~QEntity() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt53DCore.so.5
#5  0x00007fffb6f82816 in Qt3DCore::QAspectEngine::setRootEntity(QSharedPointer<Qt3DCore::QEntity>) ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt53DCore.so.5
#6  0x00007fffb6f82afa in Qt3DCore::QAspectEngine::~QAspectEngine() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt53DCore.so.5
#7  0x00007fffb6f82ce9 in Qt3DCore::QAspectEngine::~QAspectEngine() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt53DCore.so.5
#8  0x00007fffb6a00322 in Qt3DExtras::Qt3DWindow::~Qt3DWindow() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt/lib/libQt53DExtras.so.5
#9  0x00007fffb6cba3ec in Qt3DWindowWrapper::~Qt3DWindowWrapper() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/Qt3DExtras.abi3.so
#10 0x00007fffd4254fe9 in PySide::destructionVisitor(SbkObject*, void*) ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/libpyside2.abi3.so.5.12
#11 0x00007ffff4a8c722 in Shiboken::BindingManager::visitAllPyObjects(void (*)(SbkObject*, void*), void*) ()
   from /home/dolica/.local/lib/python3.6/site-packages/shiboken2/libshiboken2.abi3.so.5.12
#12 0x00007fffd4255821 in PySide::destroyQCoreApplication() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/libpyside2.abi3.so.5.12
#13 0x00007fffd42569e3 in PySide::runCleanupFunctions() ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/libpyside2.abi3.so.5.12
#14 0x00007fffce5ef1b5 in SbkQtCoreModule___moduleShutdown(_object*) ()
   from /home/dolica/.local/lib/python3.6/site-packages/PySide2/QtCore.abi3.so
#15 0x000000000056246e in PyCFunction_Call () at ../Objects/methodobject.c:112
#16 0x000000000057c2fe in PyObject_Call () at ../Objects/abstract.c:2261
#17 0x00000000005d146d in atexit_callfuncs () at ../Modules/atexitmodule.c:89
#18 0x0000000000643114 in call_py_exitfuncs () at ../Python/pylifecycle.c:1503
#19 Py_FinalizeEx () at ../Python/pylifecycle.c:568
#20 0x0000000000643268 in Py_Exit (sts=sts@entry=0) at ../Python/pylifecycle.c:1561
#21 0x0000000000643340 in handle_system_exit () at ../Python/pythonrun.c:626
#22 0x000000000064336c in PyErr_PrintEx () at ../Python/pythonrun.c:636
#23 0x000000000064375b in PyErr_Print () at ../Python/pythonrun.c:532
#24 PyRun_SimpleFileExFlags () at ../Python/pythonrun.c:425
#25 0x000000000062b26e in run_file (p_cf=0x7fffffffdd9c, filename=<optimized out>, fp=<optimized out>)
    at ../Modules/main.c:340
#26 Py_Main () at ../Modules/main.c:810
#27 0x00000000004b4cb0 in main (argc=2, argv=0x7fffffffdf98) at ../Programs/python.c:69
```

rerpha, 2019-07-09:
I wonder if this is to do with keeping the variables in python as `self` members. Perhaps deleting those variables in the constructor of our code would solve this?

rerpha, 2019-10-07:
Closed in #432

PR #432 (2019-08-13): 377: Fix crash on exit (segfault)
### Issue

Closes #377 

### Description of work

Fixes the segfault caused by python not clearing stuff up correctly with qt3d. 

Deleted the entities before qt3d tries to reference them to draw the next frame in between closing the application. 

### Acceptance Criteria

- The program exits with  exit code 0 instead of `SIGSEGV`

### UI tests

N/A

### Nominate for Group Code Review

- [ ] Nominate for code review
closed 2019-10-07 by rerpha