ess-dmsc/nexus-constructor #380: Allow multiple selection of fields

rerpha, 2019-07-09:
Currently only one field can be selected. I have tried this and it seems like it's a bit more involved than just applying a `QAbstractItemView::SelectionMode` (for example `ExtendedSelection`, which is what we want)

rerpha, 2019-07-09:
this is blocked by #359

rerpha, 2019-07-22:
how important is this? is it likely that components will have lots and lots of fields? 

if not, i don't think this should be a priority.

matthew-d-jones, 2019-08-08:
We will have a better idea whether this is important once we start using the application properly, which should be soon. Marking this low priority for now.

kmurica, 2021-02-23:
@matthew-d-jones Any news on the importance of this issue? I am a bit unsure what the benefit is here with the functionality of selecting multiple fields simultaneously. Although, my experience with the nexus constructor so far is limited. :)

matthew-d-jones, 2021-02-24:
Possibly useful when deleting multiple fields, but I'm not sure how often you'd want to do that. I think it best to close the ticket.

closed 2021-02-24 by matthew-d-jones