ess-dmsc/nexus-constructor #397: Render cylinders as cylinders in the 3D view

matthew-d-jones, 2019-07-11:
Instead of generating a mesh for cylinders, use the higher level interface of qt3d and simply tell it to render a cylinder.

matthew-d-jones, 2019-07-12:
blocked by #334

rerpha, 2020-03-30:
I suspect this will bring pretty big performance benefits for doing things like generating detector tubes and so on as it looks like most of the time the program spends when generating the mesh is the cylinder to OFF mesh conversion. We would need to add a transformation after the mesh has been generated to retain the axis direction which is where this might get a bit complex, because of the way we are storing these in the 3d view currently in a dictionary. For me replacing the `off_shape` return with a [`QCylinderMesh`](https://doc.qt.io/qt-5/qt3dextras-qcylindermesh.html) works fine but is not rotated correctly. I believe these are also instance rendered, as you can specify the number of cylinders. It's also all done in OpenGL by the looks of things so will be much quicker than raw python.

closed 2021-11-08 by ebadkamil