ess-dmsc/nexus-constructor #404: Add UI tests for main window

DolicaAkelloEgwel, 2019-07-16:


rerpha, 2019-07-17:
We should have a think about what we want to test for the main window, a couple of ideas from me are: 

- file dialogs opening when the actions/buttons in the `file` menu are selected
- pressing the nexus file layout tab button displays the silx treeview
- buttons that should be enabled in the toolbar for the component view
- expanding an item in the components

i can't really think of a good way to test the 3D view, maybe if there's an easy way to see if the sample shape is loaded, but i think this is already covered in other unit tests?

rerpha, 2019-07-26:
I don't think we're able to do this for now as the 3d window even if overridden causes pytest to segfault. Marking as blocked for now.

DominicOram, 2019-07-29:
See if this is fixed after https://github.com/ess-dmsc/nexus-constructor/issues/377

rerpha, 2019-10-09:
Still an issue, even in pyside2 5.13.1.

DominicOram, 2019-12-11:
`main_window.py` is growing and some of the logic in there is quite horrible (see `set_button_state`). If this ticket is blocked for the foreseeable future can we change this ticket to instead refactor `main_window.py` to separate out the parts that are testable and then just test them?

kmurica, 2021-02-23:
@matthew-d-jones I am curious if we can't just close this ticket? It seems obsolete. If tests are needed to be added after some additional refactoring and/or GUI development, we should create new tickets in Jira or issues here.

matthew-d-jones, 2021-02-24:
Agreed

closed 2021-02-24 by matthew-d-jones