ess-dmsc/nexus-constructor #412: Improve UI when editing floats in array fields

matthew-d-jones, 2019-07-23:
The behaviour when editing floats in an array field table is awkward, for example try typing `1.2` and you get `1.0.2`.

rerpha, 2019-07-26:
I think this is to do with the way we are updating the model and view every time the text is changed (not finished editing).

we have a couple of options here that i can think of:
- implement a "done editing" button
- revert to having to finish the text editing (lose focus, press enter/tab) 
- inherit from QLineEdit and create a line edit with our own signals? but this seems a bit overkill.

matthew-d-jones, 2019-08-19:
I think the model update on every change was only added to avoid losing data when the window is closed without deselecting the current cell. Perhaps the window close method can do the equivalent of a "done editing" button? Having a "done editing" button as well may not be a bad idea.

matthew-d-jones, 2019-10-01:
Closed by #448

PR #448 (2019-10-01): Fixing floating point array field UI issues 
### Issue

Closes #412 

### Description of work

Quick and easy fix - fixes the weird behaviour when typing doubles in the array fields editor. This was due to the validator validating every time a character was entered, and rounding up to the nearest integer. To solve this, I have connected the `editingFinished` signal of the lineEdit to the validate function. This works even if the window is closed as focus is lost. 

### Acceptance Criteria 

Test entering floating point values as a `double` type in the array fields editing box. Previously this  [... cut]
closed 2019-10-01 by matthew-d-jones