ess-dmsc/nexus-constructor #443: Control over placeholder geometry for sample, monitor

matthew-d-jones, 2019-09-07:
Rod? or common sample shapes?

kmurica, 2021-02-23:
@matthew-d-jones could you add additional information in the description of this issue? Not 100 % sure if I understand it.

matthew-d-jones, 2021-02-23:
Sorry this is a particularly badly written ticket. When a component is added and no explicit shape is defined for it then it is rendered as a cube ("placeholder shape") in the 3D view so that its position can be seen. I think there was a suggestion to allow different shapes to be used as placeholders but I don't think it was well thought through. There is another ticket to display the currently selected component in a different colour which seems like a better way to distinguish the components in the 3D view.
I will close this ticket, if it turns out to be important then the idea will resurface.

closed 2021-02-23 by matthew-d-jones