ess-dmsc/nexus-constructor #478: Improve performance of 3D view for many pixels

matthew-d-jones, 2019-10-10:
Performance seems to be fine up to around 20,000 pixels on my machine, but it does not scale linearly and rapidly becomes unusable above that.

A couple of things to consider:
- instanced rendering (QML example: https://github.com/GarageGames/Qt/tree/master/qt-5/qt3d/examples/qt3d/bigscene-instanced-qml)
- combine pixels into a single entity per detector, helps if the problem is too many entities but doesn't help if the rendering process is the bottleneck.

DominicOram, 2019-10-22:
There seems to be two bottlenecks here:

* When adding the material to the entity. You can see this by comparing the time taken to add for example the gnomon:
```
Creating entity: 5.400000000044258e-06 
Adding comp <PySide2.Qt3DExtras.Qt3DExtras.QCylinderMesh object at 0x0000015B5FD8EB48>: 6.800000000084516e-06
Adding comp <PySide2.Qt3DExtras.Qt3DExtras.QPhongAlphaMaterial object at 0x0000015B5FD96108>: 5.2999999999858716e-06
Adding comp <PySide2.Qt3DCore.Qt3DCore.QTransform object at 0x0000015B5FD8E888>: 5.300000000096894e-06
```
to that of adding something from the file:
```
Creating entity: 3.7000000077114237e-06 
Adding comp <nexus_constructor.off_renderer.OffMesh object at 0x0000015B6DD93508>: 3.4999999911633495e-06
Adding comp <PySide2.Qt3DExtras.Qt3DExtras.QPhongAlphaMaterial object at 0x0000015B5FD96948>: 0.00021510000001967455
Adding comp <PySide2.Qt3DCore.Qt3DCore.QTransform object at 0x0000015B6F25F188>: 6.100000007336348e-06
```
* Setting the parent for all the entities takes ~3 seconds for them all

DominicOram, 2019-10-23:
By creating a new material, just for the component I can bring the time down a little. Time taken with existing material:

```
Setting parent: 3.3599037999999997
Full added component: 4.781418200000001
INFO:root:Qt mesh built
Setting parent: 3.012792299999999
Full added component: 3.5747183999999983
```

Time taken with new material:
```
Setting parent: 2.3309780999999994
Full added component: 2.4645175000000004
INFO:root:Qt mesh built
Setting parent: 2.7188096
Full added component: 3.6131337000000006
```

PR #492 (2019-10-24): 478 improve performance
### Issue

Closes #478

### Description of work

Moves copying of mesh into various positions into the OFF renderer, greatly improving performance.

### Acceptance Criteria 

* Import a nexus file with a large number of pixels, confirm displayed correctly and promptly
* Add a detector with a large number of pixels, confirm displayed correctly and promptly

### UI tests

Functionality unchanged, added a bunch of unit tests on new logic.

### Nominate for Group Code Review

- [ ] Nominate for code review
closed 2019-10-24 by rerpha