ess-dmsc/nexus-constructor #495: Windows: It is not obvious which component has been selected

DominicOram, 2019-10-24:
On windows, if I click on a component in the component tree the edit/delete buttons become enabled but there is no indication which component has been selected, in the example below `select_me` has actually been selected:

[image]

rerpha, 2019-11-18:
no idea how to fix this to be honest. I have tried setting custom stylesheets, adding conditionals in the paint() method of ComponentEditorDelegate, setting mouse tracking on the treeview itself, but nothing has worked.

rerpha, 2019-11-26:
Have found a fix for this. Will push as soon as I have Internet.

PR #549 (2019-11-27): Sorting out selection visibility issue on windows
### Issue

Closes #495 

### Description of work

Highlights the selected item on windows. not sure how it looks on other platforms so you guys will have to show me. 

### Acceptance Criteria 

items are selected now. not sure why it didn't work out of the box on windows, probably a bug in QT. 

### UI tests

Not really easily testable in UI tests especially as it only affects windows 

### Nominate for Group Code Review

- [ ] Nominate for code review
closed 2019-11-27 by rerpha