ess-dmsc/nexus-constructor #498: Load V20 JSON file

matthew-d-jones, 2019-10-28:
When trying to load a V20 JSON file it currently creates a few meshes and then seems to get stuck doing something computationally intensive. I've tried leaving it for ~5 minutes, but nothing changes.

Ask @matthew-d-jones for a V20 JSON file to test with.

rerpha, 2019-11-04:
when this is loading properly lets create a bug fix release.

rerpha, 2019-11-04:
works with forwarder commands - successfully generated a forwarder command from the JSON

rerpha, 2019-11-14:
some issues i've had: 

pixel data does not get added to 3d view on file load - #527 
components are not editable and currently result in exceptions being thrown to do with not having a group - #532

rerpha, 2019-11-22:
Ok, it all seems to work so long as you edit the detector and give it a valid local file to load. I am going to close this seeing as #522 is nearly finished and completes this.

closed 2019-11-22 by rerpha