ess-dmsc/nexus-constructor #550: Remove chunk settings code (and related cruft)

SkyToGround, 2019-11-27:
We are re-writing the file-writer to not use nested structures for its component settings. For example for setting the chunk size and and log interval. Hence we need to remove this part of the code from the nexus-constructor as well. This should also fix a few bugs for us.

rerpha, 2019-11-29:
Marking as blocked as this has not yet been implemented in the filewriter. Once it has been implemented and the documentation for the commands have been updated this can be done.

rerpha, 2019-12-16:
the code in #599 can be removed when this has been done

rerpha, 2020-03-05:
Is there a jira ticket @SkyToGround ? or has it already been done?

SkyToGround, 2020-03-05:
@rerpha I do know that there is a JIRA issue for fixing this in the file-writer but I do not remember the state of it. I do not remember the state of this in the nexus-constructor either. I also do not feel I have time to look at it right now. Hence, I suggest we leave this ticket as is and I will get back to this after we are done with the re-write of the file-writer. Unless you want to work on it of course.

rerpha, 2020-03-05:
Found it: [DM-1719](https://jira.esss.lu.se/browse/DM-1719)

rerpha, 2020-08-05:
Closing, the filewriter already uses dot notation instead of nested groups, the ticket is just for updating the fw docs to reflect this.

closed 2020-08-05 by rerpha