ess-dmsc/nexus-constructor #573: V20 file created empty topic and crashed filewriter

rerpha, 2019-11-29:
When using the file generated from the nexus-constructor, the filewriter crashed because a topic field was blank. We should find out what caused this and see if we can reproduce it then fix it.

rerpha, 2019-12-16:
create a filewriter issue as well - this is an issue in both. an empty string should not cause it to segfault/crash

kmurica, 2021-02-23:
Might become a non-issue with https://github.com/ess-dmsc/nexus-constructor/issues/840 if the problem is only for file writing from nexus constructor?

matthew-d-jones, 2021-02-25:
The problem was in the output JSON, the direct Filewriter control wasn't used. However the entire data model and JSON writing code has been replaced since this ticket was created, so I will close it.

closed 2021-02-25 by matthew-d-jones