ess-dmsc/nexus-constructor #580: Can't seem to edit components with array fields

DolicaAkelloEgwel, 2019-12-04:
```
Traceback (most recent call last):
  File "/home/dolica/nexus-constructor/nexus_constructor/array_dataset_table_widget.py", line 128, in data
    value = self.array[index.row()][index.column()]
IndexError: invalid index to scalar variable.
```

matthew-d-jones, 2019-12-05:
@DolicaAkelloEgwel can you give any more details on how to reproduce this?

On `master` I am able to create a component with an array field with type double, then edit the component and edit array values with no problem.

DolicaAkelloEgwel, 2019-12-05:
I think it might be a problem specifically with 1D arrays...

matthew-d-jones, 2019-12-05:
Ah yes, thank you. The error is displayed when the "Edit" button is clicked on a field with a 1D array value, and the array editor interface is not populated with the array values.

PR #577 (2020-01-15): 540: Make disk chopper with attributes
### Issue

Closes #540
Closes #580 

### Description of work

- Allows you to set the units for the radius, slit height, and slit edges of a disk chopper using attributes in the Add Component Window. The Disk Chopper should also be completely editable.
- Also added some tests for the `ArrayDatasetTableWidget` in the process of figuring out #580.
- The functions for validating units have now been moved to their own file so that they can be shared with the `UnitValidator` and the `NexusDefinedChopperChecker`.
- Units input for the chopper attributes should  [... cut]
closed 2020-01-15 by matthew-d-jones