ess-dmsc/nexus-constructor #607: Do not use HDF5 as a data model in the nexus-constructor

SkyToGround, 2020-01-09:
Due to differences between how Python objects HDF5 paths/datasets work, it turns out that using an in memory HDF5 file as a model of the data shown in the GUI is problematic.

For this reason, we should remove the use of an in memory HDF5 file as a data model in the nexus-constructor.

rerpha, 2020-03-16:
### First steps

#### 1
- Disable any code that loads from or saves to a nexus file 
- Disable the silx tree view
- Disable IDF loading (We should enable this when we can load from a nexus file again, and use the same code to parse the output from the NexusBuilder that is used to convert IDF to HDF)

#### 2
- Modify the "Component" class with the use of the `attr.s` decorator 
 - Currently these are properties, but if they're going to be python types we can shorthand them by using `attr.s` instead. 
 - Think about what needs to be here - it should be able to hold a list of members in the component group, which in turn could be Components. We may be able to do #561 whilst we are at it. 
- Shape needs to be converted to a python model so we can use cylinders and mesh types, there are classes for this already but they're using a HDF shape group as the model and need to be converted to python

#### 3
- Modify the Transformations classes, like above. 
 - The depends_on chain implementation may be easier? we will probably still need a way of telling which transformation is depended on for the 3d view to work, and in which case we shouldn't add this to the JSON that gets saved/sent

#### 4
- Create an "Attributes" container for both components and transformations to use. This could include commonly used attributes such as `NX_class`

#### 5
- Convert the entry and instrument to the Python model
- Convert the treeview model to use the Python objects  
- Make the add component window return a component object which will be displayed in the tree view 

#### 6
- Implement a way of generating a HDF file from the model
 - This will be used for saving to a nexus file
 - Will also be used for displaying the silx nexus tree structure
- Implement a way of loading FROM a HDF file into the model. Do we still want to be able to load from nexus files? I think we do, as it means users can load up an existing file after a run and make sure it looks correct, but this is up for debate 
 - The Mantid IDF loading relies on a nexus file as an output so it'll need to be loaded into the model this way

#### 7 
- Fields will need to be converted - Array and scalar will be easy, as we now need to just return their values (rather than the dataset they are stored in), links and streams are going to be more difficult. Streams may just be a case of returning a "Stream" object that contains all of the members a stream could possibly hold. 

### Things I think this will improve: 
- Stuff like renaming will be easier, although we will still need to call all of the signals we currently do to update the Qt views 
- Sending the nexus_structure will be easier, as well as saving to JSON, as it'll just be sending the model (with certain fields blacklisted if they are added by the NC) 
- Nested groups will be easier as all that will be returned is a component object. Currently it is possible but not implemented. 
- Treeview models will be easier to read and their implementations will be easier to make sense of
- Model data to Qt display conversion will be easier if native, unchanged if using numpy. 

### Things I think this change will be detrimental towards: 
- Performance for the nexus file treeview isn't going to be as good as we'll have to keep generating a nexus file to display. As well as this, selecting links will most likely be harder as the silx tree view cannot be used easily.
- Validating fields like links is going to be tricky because they can be anywhere in the file 
- JSON by default allows for [duplicate keys](https://stackoverflow.com/questions/21832701/does-json-syntax-allow-duplicate-keys-in-an-object) which is going to cause issues when converting to HDF.
- Unless we use numpy, performance isn't going to be as good as native HDF5 for the model. If we do use numpy, we are still going to have to convert the data to human-readable qt types to be displayed and written to. 
- The standard JSON library in python does not allow for easily converting numpy arrays to JSON, and numpy types have to be converted to their respective native python types in order to be converted.
- With raw python as the model, the program is more complex - if we still want to open and write nexus files we are just moving the functionality from the model elsewhere. In my opinion this just means there are two places something could potentially go wrong.

matthew-d-jones, 2020-03-17:
1. Disable the component treeview?

4. Can't we just use a dictionary for attributes? What would the custom container do?

5. Given that they should be more simple maybe entry and instrument would be best to tackle first, before component? 

If generating HDF file is too slow for silx view then could add a button to refresh it rather than use the existing qt signal. Can eventually replace with our own treeview.

rerpha, 2020-03-17:
I have created the branch `model_change` to work off. I will also create a milestone and split this task up into sub-tickets.

SkyToGround, 2020-03-26:
@rerpha My response to your comments about the downsides to using Python for modelling the data:

1. There is an easy solution to this. Simply keep re-generating the HDF file in a background thread but only connect it to the silx tree view when the user selects that tab. That would also (potentially) help with the current latency when opening/selecting the tree view tab.
2. You might be right about that but I am not sure that I see the issue.
3. I do not see this as more of a problem than previously as Python dictionaries does not support duplicate keys.
4. Repeat after me: "Premature optimisation is the root of all evil." 😊 In short, I do not believe this to be an issue until someone has shown it to be an issue. If it is, then there are easy solutions to it.
5. This is an issue and it is an issue in the current implementation as well.
6. Well, we actually already have two places where things can go wrong (and they do) as the application can currently (actually barely) both open and create both JSON files and HDF5 files so this is no different to what it currently does. The actual difference would be that by not using HDF5 as a data model, the code for translating the NeXus structure to a Python model could be put in one place (the file read/write code) instead of being spread across the UI-code. Furthermore, I would actually suggest that we for now ignore/drop the ability of the application to open HDF5 files for now as we do not actually need it (the ESS at least) right now.

rerpha, 2020-04-20:
OK, we just had a meeting to discuss the best way of tackling the model. It seems the easiest way is to continue trying to mirror the structure used in the file-writer's JSON commands (not the flatbuffers part, just the nexus_structure field), rather than using a dictionary for any child members of a component and then dealing with conversion later on when we come to save to JSON. 

This will work ok in the implementation in #695 however some extra work may be needed when editing streams as they do not have a `name` field, which the helpers use when searching through any `children` of objects. We will need to deal with these objects either by adding a non-default handler when serialisating to JSON, or keeping a `name` field in the stream object that is not used by the file-writer.

rerpha, 2020-04-23:
Run vulture before merging `model_change` into the master branch as there will probably be some unused code that we haven't spotted.

matthew-d-jones, 2021-02-04:
done and main branch set back to `master`

closed 2021-02-04 by matthew-d-jones