ess-dmsc/nexus-constructor #629: HTTP warning

rerpha, 2020-02-13:
I don't think this is our fault and rather the nexus documentation's, but every time the component type is changed in the add/edit component window this warning occurs: 

```
js: Mixed Content: The page at 'https://manual.nexusformat.org/classes/base_classes/NXaperture.html' was loaded over a secure connection, but contains a form that targets an insecure endpoint 'http://www.google.com/search'. This endpoint should be made available over a secure connection.
js: Mixed Content: The page at 'https://manual.nexusformat.org/classes/base_classes/NXattenuator.html' was loaded over a secure connection, but contains a form that targets an insecure endpoint 'http://www.google.com/search'. This endpoint should be made available over a secure connection.
js: Mixed Content: The page at 'https://manual.nexusformat.org/classes/base_classes/NXbending_magnet.html' was loaded over a secure connection, but contains a form that targets an insecure endpoint 'http://www.google.com/search'. This endpoint should be made available over a secure connection.
js: Mixed Content: The page at 'https://manual.nexusformat.org/classes/base_classes/NXcollimator.html' was loaded over a secure connection, but contains a form that targets an insecure endpoint 'http://www.google.com/search'. This endpoint should be made available over a secure connection.
js: Mixed Content: The page at 'https://manual.nexusformat.org/classes/base_classes/NXflipper.html' was loaded over a secure connection, but contains a form that targets an insecure endpoint 'http://www.google.com/search'. This endpoint should be made available over a secure connection.
```

rerpha, 2020-02-13:
looks like it might be coming from here? https://github.com/nexusformat/definitions/blob/master/manual/source/_templates/google_search.html

zjttoefs, 2020-02-13:
Oops. Misfire of the commit message.

zjttoefs, 2020-02-13:
Would indeed expect this to be fixed now. But too lazy to test. Leaving this for you to close @rerpha :)

rerpha, 2020-02-14:
that's fixed it :+1:

closed 2020-02-14 by rerpha