ess-dmsc/nexus-constructor #637: Add validator for attrs that disallows duplicate attr names as well as `units`

rerpha, 2020-02-19:
Currently nothing is being done to validate the line edit where you specify an `attr` name. A user could enter duplicate `attrs` because of this so we should add a validator to prevent.

ebadkamil, 2021-03-11:
@matthew-d-jones is this issue related to Add fields in Add/Edit component. Because as of now we cannot have duplicated field names. So is it still an open issue?

matthew-d-jones, 2021-03-12:
@ebadkamil This ticket is about "attributes" rather than "fields": the interface is accessed by the "Attrs" button at the far right of the field controls. "units" is a special case of an attribute that is important enough to have its own control in the field controls, and so should not be allowed to be duplicated in the attributes interface.

Things that can be added as attributes are indicated with "@" symbol in the NeXus documentation. "units" is a special case which should be present on any field which is a physical quantity.

closed 2021-11-09 by mattclarke