ess-dmsc/nexus-constructor #719: Unclear which icons are greyed out

rerpha, 2020-04-22:
They are all grey besides the delete button, so we should either add some colour or raise the buttons that are currently selectable.

rerpha, 2020-04-22:
related to #720 as that also revolves around the icons being changed.

kmurica, 2021-02-23:
If colors and icons are kept as is in https://github.com/ess-dmsc/nexus-constructor/issues/720 (i.e. grey), I believe simply raising the buttons should be a sufficient approach for the user to distinguish active and inactive buttons.

ebadkamil, 2021-03-16:
Raising the button issue was already implemented  in #851 . However adding colors to buttons should remain as an open issue  as suggested by @matthew-d-jones . So not closing this issue as of now.

PR #853 (2021-03-17): ECDC-2397: Change toolbar button background color if enabled
With #851 and current PR, we can close #719
closed 2021-03-17 by kmurica