ess-dmsc/nexus-constructor #722: Add a "show button labels" option in a "view" menu

rerpha, 2020-04-22:
We should add a view menu next to "file" containing a "show button labels" tickbox that shows the labels showing what the buttons are for above the component view.

kmurica, 2021-02-23:
[image]

matthew-d-jones, 2021-02-23:
Correct.
Tickets #719 and #720 are closely related. I think this ticket is really a suggestion for part of a solution for #720.
It would be worth looking at how other software deals with this situation (toolbar where it is difficult to convey the meaning of buttons with icons alone). Perhaps the ribbon in MS Office software provides some inspiration for how text and buttons can be used together to help?

ebadkamil, 2021-02-23:
Thanks for the explanation Matt and Kenan. I am assigning this to myself to get started with `nexus_constructor`

closed 2021-03-16 by ebadkamil