ess-dmsc/nexus-constructor #734: Fix get_component_list() required for tree view

rerpha, 2020-04-30:
Currently just returns the sample. 

We should think about how this would work for nested components as they will need to be supported in the future.

rerpha, 2020-05-01:
Yesterday evening I had a bit of a poke around the codebase to see how we'd do this. I think we should be able to get away with removing the call to this in both the tree view AND the 3d view - the method will still be needed when generating new component names, but this could be more dynamic and just search for names each time instead to remove the method completely. However, i think the simplest approach for tackling this is just to fix it as it is so everything looks familiar and there are no huge design changes.

matthew-d-jones, 2020-05-28:
Done

closed 2020-05-28 by matthew-d-jones

rerpha, 2020-05-28:
Closed in #746