ess-dmsc/nexus-constructor #836: Empty value field in transformation causes error when saving to JSON

matthew-d-jones, 2021-01-11:
A verifier highlights the empty value field in red, but if the user ignore this and tries to save to JSON then an uncaught exception is raised.

kmurica, 2021-02-23:
@matthew-d-jones question regarding this issue: what would be the expected behavior? Disabling the save button or just logging the empty value as a warning in a log/printout and saving whatever is there with a dummy value for the transformation?

matthew-d-jones, 2021-02-23:
Feedback to the user should be through the GUI, not the logs. Just disabling the save button leaves the user to figure out what it wrong, so I'm not keen on that either. I would suggest displaying an error modal window when the user tries to save. It should give the user enough information to find and correct the problem/s. I don't know if there is any example of this in the "JSON saving" process already, but there is in the "JSON loading".

kmurica, 2021-02-23:
Thanks @matthew-d-jones, that really clarifies it.

ebadkamil, 2021-03-01:
simply displaying the traceback of the error message caught while saving to json, in the Error window won't be sufficient, right? It should point which field is missing.

closed 2021-11-09 by mattclarke