ess-dmsc/nexus-constructor #844: Bug in handling warnings in JSON loading logic

matthew-d-jones, 2021-02-26:
There is a bug on the following lines where strings are returned from the transformation_reader or shape_reader and appended to a list of Exceptions. Note, use of mypy (ticket #712) would have prevented this bug being introduced.

https://github.com/ess-dmsc/nexus-constructor/blob/9719bd70996a9e74c16bed92aa5f1f922f631ad2/nexus_constructor/json/load_from_json.py#L367

https://github.com/ess-dmsc/nexus-constructor/blob/9719bd70996a9e74c16bed92aa5f1f922f631ad2/nexus_constructor/json/load_from_json.py#L392

ebadkamil, 2021-03-01:
`shape_reader.warnings` and `transformation_reader.warnings ` are list.  

https://github.com/ess-dmsc/nexus-constructor/blob/9719bd70996a9e74c16bed92aa5f1f922f631ad2/nexus_constructor/json/shape_reader.py#L57

So shouldn't `self.warnings += transformation_reader.warnings` be ok?

matthew-d-jones, 2021-03-01:
Values in the list are assumed to have a `message()` method here: https://github.com/ess-dmsc/nexus-constructor/blob/9719bd70996a9e74c16bed92aa5f1f922f631ad2/nexus_constructor/main_window.py#L165
So appending `List[str]` on those lines causes a problem.

ebadkamil, 2021-03-01:
Oh Thanks,

ebadkamil, 2021-03-01:
just realised it's the same Bug reported by @kmurica  in https://github.com/ess-dmsc/nexus-constructor/issues/843. I will close this one here

closed 2021-03-01 by ebadkamil