ess-dmsc/nexus-constructor #1033: Only partial `depends_on` support

g5t, 2023-07-05:
## Background
I would like to use [`eniius`](https://github.com/mducle/eniius.git) to produce file-writer compatible JSON formatted NeXus structure information from a [`McStas`](https://mcstas.org) instrument file.

At the moment `eniius` does not correctly-handle the affine transformations which place each NXobject base class instance in an NXinstrument. Namely, the object [dependency link is missing](https://github.com/mducle/eniius/issues/11), which makes it impossible for, e.g., `nexus-constructor` to work out where components should be placed.
The [NeXus Design documentation](https://manual.nexusformat.org/design.html#coordinate-transformation-field-and-attributes) indicates that the `depends_on` link can be `.`, for no dependency, the name of a field within the object (`name`), the relative name of a field inside of a contained group (`transformations/name`), or an absolute location `/the/absolute/path/name`.
The easiest of these to implement in `eniius` is the relative name option.

## Problem
Unfortunately, the JSON reader in `nexus-constructor` only attempts to parse absolute `depends_on` locations, as is seen in `get_component_and_transform_name`:

https://github.com/ess-dmsc/nexus-constructor/blob/7808e7c8df46fce21ce02b384d8f4d9bdcbffcce/nexus_constructor/json/transformation_reader.py#L80-L88

## Desired solution
Full support of the NeXus format specification for the `depends_on` field in the `nexus-constructor` JSON reader, most importantly adding relative name support.

ggoneiESS, 2023-07-05:
Thanks for doing the diagnosis on this Greg, can you attach a file that has the issue I can test against?

ggoneiESS, 2023-07-17:
Will follow progress on this in Jira @ https://jira.esss.lu.se/browse/ECDC-3518 rather than Github

zjttoefs, 2023-07-18:
@ggoneiESS was this "not planned" intentional? If yes, there would be little reason to track on Jira either.

closed 2023-07-18 by ggoneiESS